neilbrown/gnubee-tools #6: THANKS!

TwoTwenty, 2019-02-09:
This is just a thank you bug, I am going to try this on my gnubee.
maybe you can work with puri.sm and rebrand libremnas, they seem like cool guys. 

Thanks again.

dr4Ke, 2019-05-19:
I have to add that you do a wonderful job @neilbrown 
Thank you very much!

Adonai, 2019-07-25:
@neilbrown I second this. Do you have donation address?

neilbrown, 2019-07-29:
Thanks for the encouragement.
No, not donation address, but thanks for the thought.

con-f-use, 2022-03-06:
Also big thanks from myself. Without your work, the Gnubee would be mostly useless today, seeing there's no other modern pre-build images around, that match the convenience and ease of use of yours. :heart: :partying_face: So BIG THANKS!

----------------

Small question that doesn't warent it's one issue. I just installed `gnubee-5.15.12-gbpc2.bin and it worked. But when I booted into recovery mode pressing the black button, immediately after the USB had done it's job, I didn't have any networking - no dhcp on blue and no static & dhcp server on black. Had to configure it manually. The readme says:

> The first network port (black) is by default configured with address 192.168.10.1. The second (blue) port is configured to use DHCP to request an address. You can use whichever of these is more convenient. To login you will need to know the root password, which is "GnuBee".

So what's the deal, there?