neilbrown/gnubee-tools #9: Confused by the new network configuration (GBPC1 & 5.1.4)

Adirelle, 2019-06-09:
I just tried a new firmware built using this repository and your version 5.1.4 of linux for the GPC1 and the network is broken.

Using the serial console, `ip link s` gives me this:
```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether 78:a3:51:ab:b5:58 brd ff:ff:ff:ff:ff:ff
3: ethblack@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 1e:b5:c1:4c:b3:da brd ff:ff:ff:ff:ff:ff
4: ethblue@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 1e:b5:c1:4c:b3:da brd ff:ff:ff:ff:ff:ff
```

As I did not follow every episodes of the story about the support of the network switch, I am a bit confused:

* Is that configuration normal ? Is is it the expected one ?
* About `swconfig`: What is a "kernel that supports swconfig" ? Should I try to build one ? How ? How will it affect the name of the network interfaces ?

Thanks for your help.

Adirelle, 2019-06-09:
Oh, and could we have an example of `/etc/network/interfaces` in both cases, please ?

neilbrown, 2019-06-10:
Yes, that is all normal.
ethblack is the black ethernet port.  ethblue is the blue ethernet port.
Both are accessed via a switch accessed through eth0, but that is mostly transparent.  As long as eth0 is UP, nothing else need be done.

I have
> `auto ethblack`
> `iface ethblack inet static`
> `   address 192.168.20.6`
> `   netmask 255.255.255.0`
> `   gateway 192.168.20.1`

in my interfaces  file.

A "kernel that supports swconfig" is a 4.4 or maybe 4.8 kernel with openWRT patches that provide internal switch support as developed by ralink.  Newer kernels has DSA (Distributed Switch Architecture) which is more transparent and gives different names to things.

Now that we have working DSA support, we should move away from swconfig.

Adirelle, 2019-06-11:
Ok. Thanks for the clarification. This could worth a note in the README.

Adirelle, 2019-06-11:
Wait... all seemed clear and all. But when I try to configure my network with:

```
# cat /etc/network/interfaces
auto lo
iface lo inet loopback

auto ethblue
iface ethblue inet static
    address 192.168.0.11/24
    gateway 192.168.0.254

# systemctl restart networking
Job for networking.service failed because the control process exited with error code.
See "systemctl status networking.service" and "journalctl -xe" for details.

# journalctl -u networking -e
...
nov. 03 18:21:53 exodus systemd[1]: Starting Raise network interfaces...
nov. 03 18:21:53 exodus ifup[1766]: RTNETLINK answers: File exists
nov. 03 18:21:53 exodus ifup[1766]: ifup: failed to bring up ethblue
nov. 03 18:21:55 exodus systemd[1]: networking.service: Main process exited, cod
nov. 03 18:21:55 exodus systemd[1]: Failed to start Raise network interfaces.
nov. 03 18:21:55 exodus systemd[1]: networking.service: Unit entered failed stat
nov. 03 18:21:55 exodus systemd[1]: networking.service: Failed with result 'exit

# ip link s
ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 78:a3:51:ab:b5:58 brd ff:ff:ff:ff:ff:ff
3: ethblack@eth0: <BROADCAST,MULTICAST,M-DOWN> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 26:1f:da:9c:b7:81 brd ff:ff:ff:ff:ff:ff
4: ethblue@eth0: <BROADCAST,MULTICAST,M-DOWN> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 26:1f:da:9c:b7:81 brd ff:ff:ff:ff:ff:ff
```

I think I miss something to bring eth0 up, but I am not sure how to do so.

Adirelle, 2019-06-11:
Ok, for some reason eth0 is not activated before ethblue, which causes `ifup ethblue` to fail.

neilbrown, 2019-06-14:
The initramfs should be setting eth0 "up" - there should be no need for you to do anything to eth0.
The error 
>  RTNETLINK answers: File exists
suggest that the address, or maybe a route, is already in use.
if you "ifdown --force ethblue" and then "ip addr; ip route" what does it show?  In particular, does it mention  192.168.0.11 at all?  or does a default route already exist?

Adirelle, 2019-06-15:
1/ Right after booting:

```
# cat /etc/network/interfaces
auto lo
iface lo inet loopback

auto ethblue
iface ethblue inet static
    address 192.168.0.11/24
    gateway 192.168.0.254
# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 78:a3:51:ab:b5:58 brd ff:ff:ff:ff:ff:ff
3: ethblack@eth0: <BROADCAST,MULTICAST,M-DOWN> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 8e:db:dd:53:94:54 brd ff:ff:ff:ff:ff:ff
4: ethblue@eth0: <BROADCAST,MULTICAST,M-DOWN> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 8e:db:dd:53:94:54 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.11/24 brd 192.168.0.255 scope global ethblue
       valid_lft forever preferred_lft forever
# ip route
#
```

Then:
```
# ifdown --force ethblue
RTNETLINK answers: No such process
# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 78:a3:51:ab:b5:58 brd ff:ff:ff:ff:ff:ff
3: ethblack@eth0: <BROADCAST,MULTICAST,M-DOWN> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 8e:db:dd:53:94:54 brd ff:ff:ff:ff:ff:ff
4: ethblue@eth0: <BROADCAST,MULTICAST,M-DOWN> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 8e:db:dd:53:94:54 brd ff:ff:ff:ff:ff:ff
# ip route
#
```

BTW, should I be concerned with two ethbl* interfaces having the same MAC address ?

Adirelle, 2019-06-15:
Adding `pre-up ifconfig eth0 up` to the interface configuration of `ethblue` seems to fix the issue, e.g:

```
iface ethblue inet static
    pre-up ifconfig eth0 up
    # ...
```

I am wondering if all this could be related to #10.