neilbrown/gnubee-tools #10: Missing modules.* files [5.1.4]

Adirelle, 2019-06-14:
It seems some files about the modules are missing in the build. Is it a configuration option ? I am using my own kernel configuration, which is based on yours but removes some unused drivers (mostly NIC & RTC drivers) and adds DM raid modules.

This gives me this kind of errors:

```
# iptables -S
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/5.1.4+/modules.dep.bin'
modprobe: FATAL: Module ip_tables not found in directory /lib/modules/5.1.4+
iptables v1.6.0: can't initialize iptables table `filter': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded.
# depmod -a
depmod: WARNING: could not open /lib/modules/5.1.4+/modules.builtin: No such file or directory
```

It also complained at build time, IIRC.

Adirelle, 2019-06-15:
Also, during the boot sequence:

```
[    7.752283] Preloading modules
modprobe: can't open 'modules.dep': No such file or directory
modprobe: can't open 'modules.dep': No such file or directory
modprobe: can't open 'modules.dep': No such file or directory
modprobe: can't ope[    7.796420] Loading other modules
n 'modules.dep': No such file or directory
modprobe: can't open 'modules.dep': No such file or directory
modprobe: can't open 'modules.dep': No such file or directory
modprobe: can't open 'modules.dep': No such file or directory
```

Adirelle, 2019-06-15:
Something is definitively wrong on the booted system:

```
# find /lib/modules/5.1.4+/
/lib/modules/5.1.4+/
/lib/modules/5.1.4+/build
/lib/modules/5.1.4+/kernel
/lib/modules/5.1.4+/modules.order
/lib/modules/5.1.4+/source
```

neilbrown, 2020-05-02:
I suspect this isn't still a problem, so I'll close this, but please reopen if you still have issues.
To answer your last question, install "binwalk" and try

` file=gnubee.bin; bs=$(binwalk -R "\x5d\x00\x00\x00\x04\xff\xff\xff\xff\xff\xff\xff\xff\x00\x18\x0d" $file | awk '/Raw sig/ {print $1}') ; if [ -n "$bs" ]; then dd if=$file bs=$bs skip=1 | lzcat | cpio -itv; fi
`
 Change the "file=" at the start as appropriate.

closed 2020-05-02 by neilbrown