neilbrown/gnubee-tools #11: Slight additions to network configuration

Adonai, 2019-07-25:
1. Should `tmfs` here be `tmpfs`?
https://github.com/neilbrown/gnubee-tools/blob/ccdfde0465c4a8f0a337055ec7268dceacd60677/initramfs/init#L79

2. What if I use systemd-networkd and don't need /etc/network/interfaces? How can I prevent them from being mounted?
https://github.com/neilbrown/gnubee-tools/blob/ccdfde0465c4a8f0a337055ec7268dceacd60677/initramfs/init#L43

neilbrown, 2019-07-29:
> Should tmfs here be tmpfs?
That is certainly what I meant to type, but it doesn't really matter. You can put any string at all there and it will have the same effect.

> What if I use systemd-networkd 

I don't know anything about systemd-network.  It would still need some sort of configuration, and I hope it would get the configuration from the standand place - /etc/network/interfaces.
If you, or anyone, works out what changes would be useful, I'll happily accept them, but I don't have time to dig into these issues at present.

traverseda, 2019-08-16:
Debian currently uses network-manager instead of systemd-networkd. I'd recommend doing what upstream is doing.

That being said, setting up systemd-networkd isn't hard: https://wiki.debian.org/SystemdNetworkd

I wouldn't want to try to debug it though.

Adonai, 2019-08-18:
If I come up with systemd-networkd configuration that can be integrated into initramfs I'll create PR. Thanks for fixing the typo.

closed 2019-08-18 by Adonai