neilbrown/gnubee-tools #13: issue flashing the latest firmware as described in readme

whysthatso, 2019-11-02:
hey,
first of all, thanks for all your work, much appreciated.
when i run `flashcp -v gnubee-5.2.8-gbpc2.bin /dev/mtd3` as suggested in the readme (on a gb pc2)
i get this result:
```
gnubee-5.2.8-gbpc2.bin won't fit into /dev/mtd3!
```
is this possibly the wrong target?

neilbrown, 2019-11-03:
What kernel are you currently running?
How big is mtd3? 
Try "grep mtdblock /proc/partitions" and "mtdinfo -a" .  Report all output you get.
What size is your gnubee-5.2.8-gbpc2.bin ?

whysthatso, 2019-11-03:
uname -a
```
Linux 3.10.14 #1 SMP Wed Mar 1 10:20:40 BRT 2017 mips GNU/Linux
```
grep mtdblock /proc/partitions 
```
  31        0      32768 mtdblock0
  31        1        192 mtdblock1
  31        2         64 mtdblock2
  31        3         64 mtdblock3
  31        4      32448 mtdblock4
  31        5      30249 mtdblock5
  31        6      26048 mtdblock6
```
mtdinfo -a
```
Count of MTD devices:           7
Present MTD devices:            mtd0, mtd1, mtd2, mtd3, mtd4, mtd5, mtd6
Sysfs interface supported:      yes

mtd0
Name:                           ALL
Type:                           nor
Eraseblock size:                65536 bytes, 64.0 KiB
Amount of eraseblocks:          512 (33554432 bytes, 32.0 MiB)
Minimum input/output unit size: 1 byte
Sub-page size:                  1 byte
Character device major/minor:   90:0
Bad blocks are allowed:         false
Device is writable:             true

mtd1
Name:                           Bootloader
Type:                           nor
Eraseblock size:                65536 bytes, 64.0 KiB
Amount of eraseblocks:          3 (196608 bytes, 192.0 KiB)
Minimum input/output unit size: 1 byte
Sub-page size:                  1 byte
Character device major/minor:   90:2
Bad blocks are allowed:         false
Device is writable:             true

mtd2
Name:                           Config
Type:                           nor
Eraseblock size:                65536 bytes, 64.0 KiB
Amount of eraseblocks:          1 (65536 bytes, 64.0 KiB)
Minimum input/output unit size: 1 byte
Sub-page size:                  1 byte
Character device major/minor:   90:4
Bad blocks are allowed:         false
Device is writable:             true

mtd3
Name:                           Factory
Type:                           nor
Eraseblock size:                65536 bytes, 64.0 KiB
Amount of eraseblocks:          1 (65536 bytes, 64.0 KiB)
Minimum input/output unit size: 1 byte
Sub-page size:                  1 byte
Character device major/minor:   90:6
Bad blocks are allowed:         false
Device is writable:             true

mtd4
Name:                           firmware
Type:                           nor
Eraseblock size:                65536 bytes, 64.0 KiB
Amount of eraseblocks:          507 (33226752 bytes, 31.7 MiB)
Minimum input/output unit size: 1 byte
Sub-page size:                  1 byte
Character device major/minor:   90:8
Bad blocks are allowed:         false
Device is writable:             true

mtd5
Name:                           rootfs
Type:                           nor
Eraseblock size:                65536 bytes, 64.0 KiB
Amount of eraseblocks:          472 (30975395 bytes, 29.5 MiB)
Minimum input/output unit size: 1 byte
Sub-page size:                  1 byte
Character device major/minor:   90:10
Bad blocks are allowed:         false
Device is writable:             false

mtd6
Name:                           rootfs_data
Type:                           nor
Eraseblock size:                65536 bytes, 64.0 KiB
Amount of eraseblocks:          407 (26673152 bytes, 25.4 MiB)
Minimum input/output unit size: 1 byte
Sub-page size:                  1 byte
Character device major/minor:   90:12
Bad blocks are allowed:         false
Device is writable:             true
```
ls -Al gnubee-5.2.8-gbpc2.bin
```
17299376  gnubee-5.2.8-gbpc2.bin
```

whysthatso, 2019-11-03:
`mtdinfo -a` is particularly helpful, it seems mtd4 or mtd6 are at least writeable options?

neilbrown, 2019-11-03:
I had forgotten that 3.10 had a very different partition table.
mtd4 - the one labeled "firmware" - is the one you want.
The kernel I provide doesn't have an 'all' partition, which is why there is an offset.
I have
mtd0 - 192K bootloader
mtd1 - 64K config
mtd2 - 64K factory
mtd3- 32448K - firmware

Total is 32M.
The rootfs and rootfs_data partitions in 3.10 overlap firmware, and are only need for that specfic style of firmware, which uses a root filesystem on the flash.

whysthatso, 2019-11-04:
yes indeed that solved it for me.
as i've also got a pc1, i flashed that too, but had to go in steps, first gnubee-pc1-4.4.174-2.bin, then gnubee-5.2.8-gbpc1.bin, because 3's partition scheme had the partition too small for 5.

great stuff, your work is very much appreciated.
i could document that, is there a good place to open a PR for your readme?

closed 2019-11-04 by whysthatso

neilbrown, 2019-11-04:
> i could document that, is there a good place to open a PR for your readme?

Here on github is the easiest place - through I accept patches in just about any form. Just fork this repository and make the changes...