neilbrown/gnubee-tools #14: RTC, thermal sensors, crypto engine, CPU clock scaling?

jjakob, 2020-01-06:
Hi, I recently got linux-5.4 built and running on the Gnubee PC2, thanks a lot for your work!
- I was looking at the MT7621 datasheet and it doesn't mention a RTC or thermal sensors and I don't see any on the board. That's unfortunate as these are very useful.
- crypto acceleration, clock scaling: are there any drivers for this part of the SoC?
Is there a way to change the speed of the CPU (overclocking)? The GnuBee site says it should be overclocked at 1.2GHz but in kmesg I see 900MHz.

neilbrown, 2020-01-19:
There is no RTC.  Getting time from the network is so easy these days, and the gnubee isn't much use without a network...  A little bit annoying maybe.

I don't think there are thermal sensors, but I haven't looked into it.

There is code floating around that claims to do crypto acceleration..
  https://github.com/ashevchuk/mtk-eip93-1
I haven't looked at it in any great detail.

Clock scaling appears to require use of undocumented interfaces.
The u-boot boot loader contains a binary blog which initializes the clocks in the MT7621.  The Linux kernel doesn't know how to change any of these clocks.
If you could find the blobs, you might be able to build a new u-boot which would use a different clock frequency ... or you might turn your machine into a brick.
L. D. Pinney <ldpinney@yahoo.com> might know more details - he is one of the original designers.

jjakob, 2020-01-19:
RTC: I added the fake-hwclock package to my system, that at least gives me sensible time after reboots before NTP takes over. I have it added in my fork's scripts.

Thermal sensors: yeah, the MT SoC datasheet doesn't mention any, so there are none, unless there are undocumented ones There are no other sensors on the board.

Thanks for the info on the crypto code, I may try to see if the patches apply on the recent kernel. Otherwise I won't bother as the code seems unfinished.
I was trying to see if I could overclock it but with that info, I'll pass.

closed 2020-01-19 by jjakob