neilbrown/gnubee-tools #16: vim: unable to open gpm console

jjakob, 2020-01-07:
This is logged to syslog/journal every time vim is started. A possible fix is adding /etc/vim/vimrc.local with:
```
set mouse=
set ttymouse=
```
(unconfirmed as I don't have Debian 9 any more)

On Buster this isn't present so it's possible a fix was commited upstream.
https://groups.google.com/forum/#!topic/linux.debian.bugs.dist/XOpcsX8ovIU

neilbrown, 2020-01-18:
This is a bug in vim.
vim assumes that if the TERM is not xterm, then there *must* be a Linux virtual console.  That isn't the case on the Gnubee - or in various other configurations.

Maybe the best fix would be to change the TERM for ttyS0 to "xterm". I  suspect that will (almost?) always be right in practice.

jjakob, 2020-01-19:
I think it's better to move this upstream. I can get the errors to stop by setting TERM=xterm on the serial console, but the problem is in other terminals as well (e.g. screen as used by screen and tmux). So there's not much we can do in the system. The above mouse settings are useful IMO so I'll include them in my changes, but they don't fix this issue, it's the same on Debian Buster.

closed 2020-01-19 by jjakob