neilbrown/gnubee-tools #17: ethblue/ethblack LEDs mode

jjakob, 2020-01-07:
Only the orange LED lights with Link/Act mode, regardless of speed/duplex. Usually one LED shows speed, the other Act or some combination thereof. For example: 1G: orange=link, green=act; 100M/10M: orange=off,green=link/act (this depends on what modes the chip can be set to). Is this fixable via SW/FW or is it a HW issue?

neilbrown, 2020-01-07:
Are you referring to the Gnubee-PC1 or the Gnubee-PC2 ?? The network LEDs seem to be wired differently.

For the PC-1:
The orange LED is fixed-function - it cannot be changed in software.  I don't know what the function is.
The green LED is software-controlled.
See  /sys/class/leds/gb-pc1:green:lan1/trigger

The file lists possible triggers,  Choose the one you want and write that name to the file.
If you don't know what a particular trigger name means, just try it and see what happens.

The "system" and "status" LEDs can be controlled in a similar way.

neilbrown, 2020-01-07:
I should have added: for the PC2,  I don't think the green LAN LEDs are wired to anything.  I certainly cannot find a way to drive them.

jjakob, 2020-01-07:
You're right. The PC2 schematic shows the green LEDs are not connected to anything.
Interestingly for P0 there is a pin on the SoC "ESW_P0_LED_1" that is not connected. P4 just has ESW_P4_LED_0 so there's no dedicated pin.
Here's the SoC:board:color mapping:
P0:P2:blue
P4:P1:black

P3:yellow is connected to the IP1001, bizzarely, its LEDs are connected to SoC GPIO/JTAG instead of IP1001 LED pins:
Orange=GPIO13=JTDO
Green=GPIO15=JTMS

jjakob, 2020-01-07:
Here's what I have on the PC2:
```
ls -la /sys/devices/platform/gpio-leds/leds/
total 0
drwxr-xr-x 6 root root 0 Jan  7 08:25 .
drwxr-xr-x 3 root root 0 Jan  7 08:25 ..
drwxr-xr-x 2 root root 0 Jan  7 08:25 gb-pc1:green:lan1
drwxr-xr-x 2 root root 0 Jan  7 08:25 gb-pc1:green:lan2
drwxr-xr-x 2 root root 0 Jan  7 08:25 gb-pc1:green:status
drwxr-xr-x 2 root root 0 Jan  7 08:25 gb-pc1:green:system
```

system and status control the two green LEDs next to the power switch. (you turn them on/off by setting trigger to default-on/none, brightness does nothing).

lan1 and lan2 do nothing so are superfluous. Perhaps completely new LEDs for the PC2 could be named gb-pc2:lan3-orange and green that control GPIO13/15?

jjakob, 2020-01-07:
I see this was discussed in https://github.com/gnubee-git/GnuBee_Docs/issues/82