neilbrown/gnubee-tools #19: Boot hangs after power cycle

jjakob, 2020-01-10:
- boot into initramfs
- shut down via power switch
- wait 2s
- power back on

Boot hangs and requires another hard power cycle:
```
U-Boot 1.1.3 (Aug  7 2017 - 16:23:41)

Board: MediaTek APSoC DRAM: 512 MB

Config XHCI 40M PLL
MediaTek SPI flash driver, SPI clock: 45MHz
spi device id: 1 2 19 4d
find flash: S25FL256S
============================================
MediaTek U-Boot Version: 5.0.1.0-6
--------------------------------------------
ASIC MT7621A DualCore (MAC to MT7530 Mode)
DRAM_CONF_FROM: Auto-Detection
DRAM_TYPE: DDR3
DRAM bus: 16 bit
Xtal Mode=3 OCP Ratio=1/3
Flash component: SPI Flash
Date:Aug  7 2017  Time:16:23:41
============================================
icache: sets:256, ways:4, linesz:32, total:32768
dcache: sets:256, ways:4, linesz:32, total:32768

 #### The CPU freq = 900 MHZ ####
 estimate memory size = 512 Mbytes

 Reset MT7530
set LAN/WAN WLLLL
(Re)start USB...
USB0:   mtk-xhci: init hccr be1c0000 and hcor be1c0020 hc_length 32
Register 300010f NbrPorts 3
Starting the controller
USB XHCI 0.96
scanning bus 0 for devices... 2 USB Device(s) found
       scanning bus for storage devices... 0 Storage Device(s) found

 No USB Storage found. Upgrade FW failed!

Please choose the operation:
   1: Load system code to SDRAM via TFTP.
   2: Load system code then write to Flash via TFTP.
   3: Boot system code via Flash (default).
   4: Enter boot command line interface.
   5: Load system code then write to Flash via USB Storage.
   6: Load system code then write to Flash via Httpd.
   9: Load U-Boot code then write to Flash via TFTP.                                                                                                                                                     0

3: System Boot system code via Flash.
## Checking image at bc050000 ...
   Image Name:   Linux-5.4.6+
   Image Type:   MIPS Linux Kernel Image (uncompressed)
   Data Size:    20723952 Bytes = 19.8 MB
   Load Address: 80001000
   Entry Point:  80686c20
   Verifying Checksum ... OK
OK
No initrd
## Transferring control to Linux (at address 80686c20) ...
## Giving linux memsize in MB, 512

Starting kernel ...

[    0.000000] Linux version 5.4.6+ (jernej@librem) (gcc version 8.3.0 (Debian 8.3.0-2)) #12 SMP Thu Jan 9 01:00:53 CET 2020
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[... 154 lines cut ...]
[   19.571976] mt7621-pci 1e140000.pcie: PCIE2 enabled
[   19.581719] mt7621-pci 1e140000.pcie: PCI coherence region base: 0x60000000, mask/settings: 0xf0000002
[   19.600450] mt7621-pci 1e140000.pcie: PCI host bridge to bus 0000:00
[   19.613127] pci_bus 0000:00: root bus resource [io  0xffffffff]
[   19.624919] pci_bus 0000:00: root bus resource [mem 0x60000000-0x6fffffff]
[   19.638609] pci_bus 0000:00: root bus resource [bus 00-ff]
[   19.649578] pci 0000:00:00.0: [0e8d:0801] type 01 class 0x0d8000
[   19.661574] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x01ffffff]
[   19.674058] pci 0000:00:00.0: reg 0x14: [mem 0x00000000-0x0000ffff]
[   19.686628] pci 0000:00:00.0: supports D1
[   19.694613] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
[   19.706454] pci 0000:00:01.0: [0e8d:0801] type 01 class 0x0d8000
[   19.718451] pci 0000:00:01.0: reg 0x10: [mem 0x00000000-0x01ffffff]
[   19.730933] pci 0000:00:01.0: reg 0x14: [mem 0x00000000-0x0000ffff]
[   19.743485] pci 0000:00:01.0: supports D1
[   19.751452] pci 0000:00:01.0: PME# supported from D0 D1 D3hot
[   19.763256] pci 0000:00:02.0: [0e8d:0801] type 01 class 0x060400
[   19.775253] pci 0000:00:02.0: reg 0x10: [mem 0x00000000-0x7fffffff]
[   19.787737] pci 0000:00:02.0: reg 0x14: [mem 0x00000000-0x0000ffff]
[   19.800287] pci 0000:00:02.0: supports D1
[   19.808272] pci 0000:00:02.0: PME# supported from D0 D1 D3hot
[   19.820926] pci 0000:00:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[   19.836876] pci 0000:00:01.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[   19.852814] pci 0000:00:02.0: bridge configuration invalid ([bus 00-00]), reconfiguring
```

neilbrown, 2020-01-10:
This is a known problem introduced when the PCI code was being cleaned up.
I think it started failing about
Linux Commit 745eeeac68d7 ("staging: mt7621-pci: factor out 'mt7621_pcie_enable_port' function")
but when I was trying to dig deeper into it, I couldn't make it fail reliably, so I found find out exactly what caused the regression.

jjakob, 2020-01-19:
I could repeat this 3 times in a row once. I think it's how I do the shutdowns: poweroff from system, then as it halts or starts rebooting in u-boot, turn off power switch, wait ~2-3s, power on again.
As I said, last time I did this, I had to hard power off 3 times as it hung on the pci init each time.

Looking at the commit you referenced, https://github.com/neilbrown/linux/commit/745eeeac68d7
I see an obvious difference in order and timing between the old and new code. Does reverting that commit fix the issue? I might try if it fixes the issue on my end (I need to get some time to assemble a 2nd Gnubee as the one I tested on is now in use)

neilbrown, 2020-05-02:
I believe this is now fixed, as of my 5.6.3 kernels (hopefully the fix will be in mainline soon).
So I'll close this bug now.
If you still have problems with 5.6.3 or later, please re-open.

closed 2020-05-02 by neilbrown