neilbrown/gnubee-tools #21: Build on gnubee generates none booting image

BlackLotus, 2020-01-30:
When building the image on the gnubee itself an image is created, but when booting it stops after "verifying checksum". Cross compiling just works.

neilbrown, 2020-01-31:
Strange ... I'll build one and see what happens.  Might take a while :-)

BlackLotus, 2020-02-01:
Just for reproducability: I tried both of the images linked here https://github.com/gnubee-git/GnuBee_Docs/wiki/Install-firmware#debian with the debian system with the procedure described here https://github.com/gnubee-git/GnuBee_Docs/wiki/Debian

neilbrown, 2020-03-27:
(Sorry for the delay).
I don't understand that last comment.  The images linked there not lot related to this gnubee-tools package ... except that they are used on the same hardware.

Or are you saying that you installed those firmware images, then used gnubee-tool running on those images to build a new firmware image?  That would be somewhat interesting, but not something that I want to spend any time on as those are very old images and I would likely just find  there is something missing there.

If you cross-compile a firmware image with gnubee-tools, then boot the gnubee and build a new image natively there - and that doesn't work.  Then I would be interested.