neilbrown/gnubee-tools #22: "Please run 'gbmake initramfs' first"

brettmilford, 2020-03-02:
Hi,
First: thanks for your hard work!

I always appear to his this clause in the `gbmake` script.
https://github.com/neilbrown/gnubee-tools/blob/master/scripts/gbmake#L194

I have an initramfs dir in the current directory but it doesn't contain the /init file, it does appear to contain everything else.

neilbrown, 2020-03-02:
The "init" script is one of the last things copied in.  It is copied from the initramfs directory of gnubee-tools.
So assuming it is still there in gnubee-tools and hasn't landed in your initramfs, it seems likely that the mkinitramfs script failed somewhere.
Can you run "gbmake iniitramfs > /tmp/log 2>&1" and report what the last few lines are?  That should help me see how far it gets.

brettmilford, 2020-03-02:
```
$ tail /tmp/log
## Adding /usr/lib/mipsel-linux-gnu/libgpg-error.so.0
## Adding /usr/lib/mipsel-linux-gnu/libnss_files.so.2
## Adding /usr/lib/mipsel-linux-gnu/libnsl.so.1
## Install script bin
## Install script etc
## Install script lib
## Install script run
## Install script sbin
## Install script usr
initramfs: 184K initramfs
```

neilbrown, 2020-03-02:
That's weird.  The last few "Install script" lines is where it should install init (and config and keep and network-scripts and udhcpc.script) but it is getting entirely the wrong content.
So presumably the "gbtools" variable has the wrong value.
That is set by:
```
prog=$0
if [ -L "$prog" ]; then prog=`readlink $prog`; fi
gbtools=`realpath $(dirname $prog)/..`

```
How, exactly, are you running "gbmake"?? do you type the full path, or is it in a 'bin' directory?  If that latter, what exactly is in the bin directory?  A symlink or a hard link or a copy or ...??

brettmilford, 2020-03-02:
So I ran the firmware command like this 
```
~/prjs/linux$ ../gnubee-tools/scripts/gbmake firmware gbpc1-5.4
```
and the initramfs command like this 
```
~/prjs/linux$ ../gnubee-tools/scripts/gbmake initramfs > /tmp/log 2>&1
```
I'm compiling this from my desktop on Ubuntu 20.04; its possible I might be using newer packages of something? I installed everything suggested in the readme, downloaded the git repos as suggested and ran the commands as above and in the readme. When it came to compiling binutils for mips I had to use the tag binutils-2_32 because of an error similar to this:   https://github.com/tkchia/binutils-ia16/issues/2 , everthing otherwise the same.

neilbrown, 2020-03-02:
Curiouser and curiouser.
At the end of "mkinitramfs" is the line
`for i in $gbtools/initramfs/*`
and that for loop seems to find directories named bin, etc, lib, run, .... which are all in '/' - and possibly elsewhere.
Maybe add "echo $i" into that for loop, and see what 'i' is getting set to.

brettmilford, 2020-03-02:
```
~/prjs/linux$ ../gnubee-tools/scripts/gbmake initramfs > /tmp/log 2>&1
~/prjs/linux$ tail -n 20 /tmp/log
## Adding /usr/lib/mipsel-linux-gnu/libgnutls.so.30
## Adding /usr/lib/mipsel-linux-gnu/libpsl.so.5
## Adding /usr/lib/mipsel-linux-gnu/libbz2.so.1.0
## Adding /usr/lib/mipsel-linux-gnu/libgcrypt.so.20
## Adding /usr/lib/mipsel-linux-gnu/libgpg-error.so.0
## Adding /usr/lib/mipsel-linux-gnu/libnss_files.so.2
## Adding /usr/lib/mipsel-linux-gnu/libnsl.so.1
## Install script bin
/home/brett/prjs/gnubee-tools/initramfs/bin
## Install script etc
/home/brett/prjs/gnubee-tools/initramfs/etc
## Install script lib
/home/brett/prjs/gnubee-tools/initramfs/lib
## Install script run
/home/brett/prjs/gnubee-tools/initramfs/run
## Install script sbin
/home/brett/prjs/gnubee-tools/initramfs/sbin
## Install script usr
/home/brett/prjs/gnubee-tools/initramfs/usr
initramfs: 184K initramfs
~/prjs/linux$ ls -lha ../gnubee-tools/initramfs/
total 48K
drwxr-xr-x  7 brett brett  8 Mar  2 10:52 .
drwxr-xr-x 10 brett brett 18 Mar  2 13:16 ..
drwxr-xr-x  2 brett brett 26 Mar  2 10:52 bin
drwxr-xr-x  4 brett brett  5 Mar  2 10:52 etc
drwxr-xr-x  3 brett brett  3 Mar  2 10:52 lib
drwxr-xr-x  3 brett brett  3 Mar  2 10:52 run
lrwxrwxrwx  1 brett brett  3 Mar  2 10:52 sbin -> bin
drwxr-xr-x  3 brett brett  6 Mar  2 10:52 usr
~/prjs/linux$
```

neilbrown, 2020-03-02:
Ahh... your gnubee-tools/initramfs directory is corrupted.
My guess is that your ran "gbmake initramfs" while in the gnubee-tools directory, or something like that.
In the "gnunee-tools" do "rm -r initramfs; git restore initramfs".  That should restore it.
If you like, you could run "git status" first, just see see what is missing and what is extra.
Depending on how new your "git" is, "git restore" might not work.  In that case try "git checkout initramfs"

brettmilford, 2020-03-02:
That appears to have done the trick, I must have messed up my commands/ done something funny early on. Thank you!

closed 2020-03-02 by brettmilford