neilbrown/gnubee-tools #26: HELP GnuBee PC1 segfaults most processes

vgiralt, 2021-01-06:
Hi, this is not really a gnubee-tools issue, but those hanging around in this repository are quite knowledgeable with GnuBee, so I decided to disturb you ...
I have a GnuBee PC1 with six drives that has been humming nicely for some years (it had an uptime of over 400 days when I had to reboot it). I had the stupid idea to dist-upgrade from stretch to buster without making an snapshot of the running LV :-( (yes, a bad move from a seasoned sysadmin with over 35 years of experience).
The buster upgraded system was segfaulting many commands and operations, so I have debootstrapped several versions of stretch, to no avail ...
Is there any way to know for sure that my system is definitely broken before I shell out some money for a replacement or hardware to connect the six drives to a RPi4?
Thanks for your time!
P.S.: bigger thanks to Neil Brown and his excellent work with the system his tools generate. I love the gnubee-config.txt idea, a total life saver!

vincele, 2022-12-03:
If it only did that after the upgrade and was "humming nicely for some years" before, I'd say the problem may not be with the HW.

Did you try to reinstall an older version ?

vgiralt, 2023-01-18:
> Did you try to reinstall an older version ?

I tried with the last properly running kernel. Still have to try a fresh stretch install ...
Thanks!

vincele, 2023-02-01:
Random process segfaulting may be an indicator of RAM problem, dunno if you can run memtest...