neilbrown/gnubee-tools #28: gbpc2-5.10 Parsing DT failed

jjakob, 2021-02-17:
Hi, I recently upgraded from 5.4 to 5.10 and after booting I noticed that the LEDs weren't working any more. Investigating, I found the leds-gpio module didn't load cause of "error -16". Looking at dmesg shows some errors in applying the DT settings. Full dmesg attached.
[dmesg_dt_failed.log](https://github.com/neilbrown/gnubee-tools/files/5999277/dmesg_dt_failed.log)

```
[   20.894408] sdhci: Secure Digital Host Controller Interface driver
[   20.906760] sdhci: Copyright(c) Pierre Ossman
[   20.917777] rt2880-pinmux pinctrl: sdhci is already enabled
[   20.928996] mtk-msdc 1e130000.sdhci: Error applying setting, reverse things back
[   20.943758] rt2880-pinmux pinctrl: sdhci is already enabled
[   20.954865] mtk-msdc 1e130000.sdhci: Error applying setting, reverse things back
[   20.969724] ledtrig-cpu: registered to indicate activity on CPUs
[   20.982070] mt7621-pci 1e140000.pcie: host bridge /pcie@1e140000 ranges:
[   20.995619] mt7621-pci 1e140000.pcie:      MEM 0x0060000000..0x006fffffff -> 0x0000000000
[   21.011985] mt7621-pci 1e140000.pcie:       IO 0x001e160000..0x001e16ffff -> 0x0000000000
[   21.028428] mt7621-pci 1e140000.pcie: Parsing DT failed
...
[   21.258441] rt2880-pinmux pinctrl: pcie is already enabled
[   21.269404] mt7621-pci 1e140000.pcie: Error applying setting, reverse things back
...
[   57.924652] leds-gpio: probe of gpio-leds failed with error -16
```

hvontres, 2021-05-01:
I can confirm that this also happens with the 5.6.14 and 5.7.2 images

hvontres, 2021-05-02:
Did some more digging. It looks like the following entry in drivers/staging/mt7621-dts/gbpc1.dts is causing the leds to fail:

> &pcie {
> 	pinctrl-names = "default";
> 	pinctrl-0 = <&pcie_pins>;
> 
> 	reset-gpios = <&gpio 19 GPIO_ACTIVE_LOW>,
> 			**<&gpio 8 GPIO_ACTIVE_LOW>,
> 			<&gpio 7 GPIO_ACTIVE_LOW>;**
> 	status = "okay";
> };

On my PC2, I put this override in drivers/staging/mt7621-dts/gbpc1.dts:
```
&pcie {
	pinctrl-names = "default";
	pinctrl-0 = <&pcie_pins>;

	reset-gpios = <&gpio 19 GPIO_ACTIVE_LOW>;
	status = "okay";
};
```
I need to look at the schematic some more, but I don't think the other two resets are hooked up on the PC2 as the pins used are part of the RGMII interface for eth1. I also think they should be GPIO22 and GPIO27 based on my reading of the schematic.

Hope this helps.

hvontres, 2021-05-02:
After looking at the schematic some more here is what seems to be going on:
GB-PC1:
reset pins: GPIO 19, GPIO  22 and GPIO 27
GB-PC2:
No Reset pins connected between the CPU and the Pcie to sata bridges