neilbrown/gnubee-tools #31: Process after firmware flash

whysthatso, 2021-11-29:
I've just upgraded on a gb1 from 5.2.8 to 5.10.1 (thanks @neilbrown for your work by the way)

i'm running debian buster on it (10.11)

the upgrade was inspired by problems using the second network port (it would for some reason not receive the dhcp ack packages from my router, and router side i would see bad checksum udp packages), and that got fixed. the machine is pretty much working, but i'm seeing some errors in the boot messages, and i was wondering if there is anything to do with debian after flashing a new kernel. i could imagine that a bunch of references get outdated or kernel configs have changed.

here's an example of the boot log output errors:

```
[   21.679144] Run /init as init process
modprobe: module ahci not found in modules.dep
[   21.778949] usbcore: registered new interface driver usbfs
[   21.790145] usbcore: registered new interface driver hub
[   21.800990] usbcore: registered new device driver usb
[   21.819823] xhci-mtk 1e1c0000.xhci: supply vbus not found, using dummy regulator
[   21.834989] xhci-mtk 1e1c0000.xhci: supply vusb33 not found, using dummy regulator
[   21.850374] xhci-mtk 1e1c0000.xhci: xHCI Host Controller
[   21.861021] xhci-mtk 1e1c0000.xhci: new USB bus registered, assigned bus number 1
[   21.881019] xhci-mtk 1e1c0000.xhci: hcc params 0x01401198 hci version 0x96 quirks 0x0000000000210010
[   21.899356] xhci-mtk 1e1c0000.xhci: irq 20, io mem 0x1e1c0000
[   21.912346] hub 1-0:1.0: USB hub found
[   21.919983] hub 1-0:1.0: 2 ports detected
[   21.928606] xhci-mtk 1e1c0000.xhci: xHCI Host Controller
[   21.939275] xhci-mtk 1e1c0000.xhci: new USB bus registered, assigned bus number 2
[   21.954233] xhci-mtk 1e1c0000.xhci: Host supports USB 3.0 SuperSpeed
[   21.967079] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[   21.984277] hub 2-0:1.0: USB hub found
[   21.991933] hub 2-0:1.0: 1 port detected
[   22.014725] usbcore: registered new interface driver usb-storage
modprobe: module sd_mod not found in modules.dep
modprobe: module mtk_sd not found in modules.dep
modprobe: module mmc_block not found in modules.dep
modprobe: module ext4 not found in modules.dep
modprobe: module gpio_keys_polled not found in modules.dep
[   22.155815] input: gpio-keys as /devices/platform/gpio-keys/input/input0
[   22.198750] mtk_soc_eth 1e100000.ethernet eth0: configuring for fixed/rgmii link mode
[   22.214759] mtk_soc_eth 1e100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   22.231682] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
ip: SIOCGIFINDEX: No such device
[   25.470458] random: crng init done
[   26.142440] Searching for partition GNUBEE-ROOT...
[   26.152184] leds-gpio: probe of gpio-leds failed with error -16
/init: cd: line 10: can't cd to /sys/class/leds/gb-pc1:green:system: No such file or directory
modprobe: module[   26.200944] Assembling md arrays
 md_mod not found in modules.dep
mdadm: No arrays found in config file or automatically
[   26.313101] device-mapper: ioctl: 4.43.0-ioctl (2020-10-01) initialised: dm-devel@redhat.com
[   26.331031] Activating LVM volumes
findfs: unable to resolve 'PARTLABEL=GNUBEE-CRYPT-ROOT'
modprobe: module fs-ext4 not found in modules.dep
[   28.336238] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)
[   28.381627] Partition GNUBEE-ROOT found. Starting...
grep: /mnt/root/etc/network/interfaces.d/*: No such file or directory
sed: /mnt/root/etc/network/interfaces.d/*: No such file or directory
/init: cd: line 84: can't cd to /mnt/root/etc/network/script.d: No such file or directory
===== Created temporary /etc/network to handle new interface names ======
ifconfig: SIOCGIFFLAGS: No such device
ifconfig: SIOCGIFFLAGS: No such device
ifconfig: SIOCGIFFLAGS: No such device
/init: cd: line 10: can't cd to /sys/class/leds/gb-pc1:green:system: No such file or directory
/init: cd: line 10: can't cd to /sys/class/leds/gb-pc1:green:status: No such file or directory
[   30.394839] systemd[1]: System time before build time, advancing clock.
[   30.438633] systemd[1]: Inserted module 'autofs4'
[   30.541345] systemd[1]: systemd 241 running in system mode. (+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD -IDN2 +IDN -PCRE2 default-hierarchy=hybrid)
[   30.584895] systemd[1]: Detected architecture mips.

Welcome to Debian GNU/Linux 10 (buster)!

[   30.643254] systemd[1]: Set hostname to <gnubee1>.
[   31.836958] systemd[1]: File /lib/systemd/system/systemd-journald.service:12 configures an IP firewall (IPAddressDeny=any), but the local system does not support BPF/cgroup based firewalling.
[   31.870987] systemd[1]: Proceeding WITHOUT firewalling in effect! (This warning is only shown for the first loaded unit using IP firewalling.)
[   32.463632] systemd[1]: /lib/systemd/system/rpc-statd.service:13: PIDFile= references path below legacy directory /var/run/, updating /var/run/rpc.statd.pid â→ /run/rpc.statd.pid; please update the unit file accordingly.
[   32.667020] systemd[1]: Listening on initctl Compatibility Named Pipe.
[  OK  ] Listening on initctl Compatibility Named Pipe.
[   32.722776] systemd[1]: Listening on Journal Socket (/dev/log).
[  OK  ] Listening on Journal Socket (/dev/log).
[   32.771791] systemd[1]: Reached target Swap.
[  OK  ] Reached target Swap.
[   32.812143] systemd[1]: Started Forward Password Requests to Wall Directory Watch.
[  OK  ] Started Forward Password Râ…uests to Wall Directory Watch.
[  OK  ] Created slice system-postfix.slice.
[  OK  ] Created slice system-getty.slice.
[  OK  ] Listening on Journal Socket.
         Starting Set the console keyboard layout...
         Mounting Kernel Debug File System...
[  OK  ] Listening on fsck to fsckd communication Socket.
         Starting File System Check on Root Device...
         Starting Nameserver information manager...
```

vincele, 2022-12-05:
Hello,

I think the modprobe errors are harmless, they are resulting from drivers as modules being linked back into the kernel. I only checked for ahci, but I assume the others are the same.

I'll PR a fix, removing the modprobe calls.

I think the led cd error has been fixed by:
https://github.com/neilbrown/gnubee-tools/commit/e783244d32a53e1ff29d3eaeb7f9ad9b36b2a0c2

I think the interfaces.d errors should be fixed by:
https://github.com/neilbrown/gnubee-tools/commit/69d42832194abd5f604790003b7dc9ae42e49d10

Which should cover it, @whysthatso, could you retry with latest image and report back here ?

vincele, 2023-02-06:
@neilbrown is the `kern_config/gbpc1-4.4` still useful ?
I'm preparing the modprobe cleanup, and all the modprobed modules were moved builtin after this version...

vincele, 2023-02-12:
I think this could be closed, almost all issues should be fixed by next build...

neilbrown, 2023-02-12:
I agree.  I've just pushed a removal of some of the older configs.

closed 2023-02-12 by neilbrown

whysthatso, 2023-02-12:
i'll try to get some feedback together as soon as i have time to tinker with the gnubees again.
they've been running the last couple of months quiet and reliably, so i didn't touch them really in favor of other projects.