neilbrown/gnubee-tools #32: PC-1 Need stable Firmware

penguinpages, 2021-12-11:
Neil And other Supporters

Thanks for all your work.  I purchase GnuBee NAS as my router vendor ...well decided to drop USB disk support and I need a dedicaed NFS export for my K8 projects as well as backups.

After building hardware, and figuring out USB Serial port issues (windows driver stupi, and gave up on UAT issues with Plex over UAT cable).   I have attempted to flash (usb key)  images of various versions to the pc1
File locations:
https://neil.brown.name/gnubee/?C=M;O=D

Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
-a---l        12/10/2021   5:26 PM       22542704 gnubee-5.10.1-gbpc1.bin
-a---l        12/10/2021   5:07 PM       15341936 gnubee-5.10.1-gbpc1u.bin
-a---l        12/10/2021  10:47 PM       20478224 gnubee-5.6.16-gbpc1.bin
-a---l        12/10/2021  10:28 PM       20605200 gnubee-5.7.2-gbpc1.bin

They have different issues. All have issue of tty working to output but not allow remote access.   I have SD disk with 8GB partition 10GB Swap and 90GB TBD...   and when I pull / insert after boot it shows on console .. so I know tty is still live and boot done.  

attached is sequence of flash and attempt to get console

I also made USB disk with gnubee-config.txt  >>
CONFIGURE_NET=yes
CONFIGURE_BLACK_IP=172.16.100.110
CONSOLE_SHELL=yes

I am trying to learn how to build this out but also need to make sure along way I understand the system and how to do root cause so I can trust it to host my data.

Questions:
1) Why are images booting without NICs up (DHCP or static)... I watch switch to see MAC .. reguardless of IP..  and these images do not show L2 into switch

2) why is console not responsive to interface

[Fail_gnubee-5.7.2-gbpc1_noNIC_noTTYinput.txt](https://github.com/neilbrown/gnubee-tools/files/7696416/Fail_gnubee-5.7.2-gbpc1_noNIC_noTTYinput.txt)

vincele, 2022-12-03:
The log shows an busybox being there.
What did you use for the serial console (gnu screen, cu, minicom...).
Maybe you can try another one.

neilbrown, 2022-12-28:
My only guess for the network not being configured properly is that the gnubee-config.txt might have a "RETURN" character (\r) at the end of each line.  This would happen if the file was created using MS-Window.
My next firmware release will strip any RETURN characters out.

If you cannot successfully type at the console, maybe the parity setting is wrong - I cannot think of anything else.

As this issue is so old (sorry!) I'll close it,  If you pursue this again and have more problems, please open another issue.

closed 2022-12-28 by neilbrown