neilbrown/gnubee-tools #33: ../arch/mips/ralink/mt7621.c:36:22: error: ‘__builtin_memcmp’ reading between 1 and 4 bytes from a region of size 0 [-Werror=stringop-overread]

jjakob, 2022-01-12:
Hi, I'm trying to compile gbpc2-5.15 on Gentoo, I followed the steps:
- installed mipsel cross-compile environment (Gentoo specific)
- cloned gnubee-tools
- set CROSS_COMPILE=/usr/bin/mipsel-unknown-linux-gnu-
- make chroot
- git clone --depth=20 https://github.com/neilbrown/linux.git -b gnubee/v5.15
- cd linux
- ../scripts/gbmake firmware gbpc2-5.15

I'm getting the following error:
```
  CC      arch/mips/ralink/mt7621.o
../arch/mips/ralink/mt7621.c: In function ‘mt7621_memory_detect’:
../arch/mips/ralink/mt7621.c:36:22: error: ‘__builtin_memcmp’ reading between 1 and 4 bytes from a region of size 0 [-Werror=stringop-overread]
   36 |                 if (!__builtin_memcmp(dm, dm + size, sizeof(detect_magic)))
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../arch/mips/ralink/mt7621.c:23:14: note: at offset 33554432 into source object ‘detect_magic’ of size 4
   23 | static void *detect_magic __initdata = detect_memory_region;
      |              ^~~~~~~~~~~~
../arch/mips/ralink/mt7621.c:42:13: error: ‘__builtin_memcmp’ reading between 1 and 4 bytes from a region of size 0 [-Werror=stringop-overread]
   42 |             __builtin_memcmp(dm, dm + size, sizeof(detect_magic))) {
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../arch/mips/ralink/mt7621.c:23:14: note: at offset 268435456 into source object ‘detect_magic’ of size 4
   23 | static void *detect_magic __initdata = detect_memory_region;
      |              ^~~~~~~~~~~~
cc1: all warnings being treated as errors
make[3]: *** [../scripts/Makefile.build:277: arch/mips/ralink/mt7621.o] Error 1
make[2]: *** [../scripts/Makefile.build:540: arch/mips/ralink] Error 2
make[1]: *** [/home/jernej/git/gnubee-tools/linux/Makefile:1868: arch/mips] Error 2
make[1]: Leaving directory '/home/jernej/git/gnubee-tools/linux/O'
make: *** [Makefile:219: __sub-make] Error 2
```

Info on environment:
```
gentoo ~ # mipsel-unknown-linux-gnu-emerge --info
Portage 3.0.28 (python 3.9.9-final-0, embedded, gcc-11.2.1, unavailable, 5.15.11-gentoo-x86_64 x86_64)
=================================================================
System uname: Linux-5.15.11-gentoo-x86_64-x86_64-Intel-R-_Core-TM-2_Quad_CPU_Q6600_@_2.40GHz-with-glibc2.33
KiB Mem:     8080604 total,    411728 free
KiB Swap:   16777212 total,  16286460 free
Timestamp of repository gentoo: Tue, 11 Jan 2022 18:00:01 +0000
Head commit of repository gentoo: 82f61656e59454131d94c763c2118f7d316e38a5
sh bash 5.1_p8
ld GNU ld (Gentoo 2.37_p1 p1) 2.37
Repositories:

gentoo
    location: /var/db/repos/gentoo
    sync-type: rsync
    sync-uri: rsync://rsync.gentoo.org/gentoo-portage
    priority: -1000
    sync-rsync-verify-jobs: 1
    sync-rsync-extra-opts: 
    sync-rsync-verify-metamanifest: yes
    sync-rsync-verify-max-age: 24

ACCEPT_KEYWORDS="mips ~mips"
ACCEPT_LICENSE="@FREE"
CBUILD="x86_64-pc-linux-gnu"
CFLAGS="-O2 -pipe -fomit-frame-pointer"
CHOST="mipsel-unknown-linux-gnu"
CONFIG_PROTECT="/etc /usr/share/config /usr/share/gnupg/qualified.txt"
CONFIG_PROTECT_MASK="/etc/ca-certificates.conf /etc/dconf /etc/env.d /etc/fonts/fonts.conf /etc/gentoo-release /etc/revdep-rebuild /etc/sandbox.d /etc/terminfo"
CXXFLAGS="-O2 -pipe -fomit-frame-pointer"
DISTDIR="/var/cache/distfiles"
FCFLAGS="-Os -pipe"
FEATURES="assume-digests binpkg-docompress binpkg-dostrip binpkg-logs buildpkg config-protect-if-modified distlocks ebuild-locks fixlafiles ipc-sandbox merge-sync multilib-strict network-sandbox news nodoc noinfo noman parallel-fetch pid-sandbox preserve-libs protect-owned qa-unresolved-soname-deps sandbox sfperms strict unknown-features-warn unmerge-logs unmerge-orphans userfetch userpriv usersandbox usersync xattr"
FFLAGS="-Os -pipe"
GENTOO_MIRRORS="http://distfiles.gentoo.org"
LANG="en_US.utf8"
LDFLAGS=""
LINGUAS="en"
PKGDIR="/usr/mipsel-unknown-linux-gnu/packages/"
PORTAGE_CONFIGROOT="/usr/mipsel-unknown-linux-gnu/"
PORTAGE_RSYNC_OPTS="--recursive --links --safe-links --perms --times --omit-dir-times --compress --force --whole-file --delete --stats --human-readable --timeout=180 --exclude=/distfiles --exclude=/local --exclude=/packages --exclude=/.git"
PORTAGE_TMPDIR="/usr/mipsel-unknown-linux-gnu/tmp/"
SHELL="/bin/bash"
USE="kdrive minimal mips multicall zlib" ELIBC="glibc" INPUT_DEVICES="evdev" KERNEL="linux" USERLAND="GNU" VIDEO_CARDS="fbdev"
Unset:  ADDR2LINE, AR, ARFLAGS, AS, ASFLAGS, CC, CCLD, CONFIG_SHELL, CPP, CPPFLAGS, CTARGET, CXX, CXXFILT, ELFEDIT, EMERGE_DEFAULT_OPTS, ENV_UNSET, EXTRA_ECONF, F77FLAGS, FC, GCOV, GPROF, INSTALL_MASK, LC_ALL, LD, LEX, LFLAGS, LIBTOOL, MAKE, MAKEFLAGS, MAKEOPTS, NM, OBJCOPY, OBJDUMP, PORTAGE_BINHOST, PORTAGE_BUNZIP2_COMMAND, PORTAGE_COMPRESS, PORTAGE_COMPRESS_FLAGS, PORTAGE_RSYNC_EXTRA_OPTS, RANLIB, READELF, RUSTFLAGS, SIZE, STRINGS, STRIP, YACC, YFLAGS
```

jjakob, 2022-01-12:
Tried it on PureOS byzantium, could not reproduce the error. There must be some difference in toolchain configuration, environment or versions.
```
root@librem:~# /usr/bin/mipsel-linux-gnu-gcc --version
mipsel-linux-gnu-gcc (Debian 10.2.1-6) 10.2.1 20210110
```

neilbrown, 2022-01-13:
It looks like newer versions of gcc are detecting code that looks odd.
Certainly that code does look odd.  What it is doing is quite sensible, but it isn't try to determine something about the memory hardware, which is not the sort of thing the compiler would normally expect to see.  It is looking to see if memory appear to "repeat" after some power-of-2, which would indicate what the usable memory size is.
We'll probably need to eventually come up with a work-around to stop the compiler from complaining, but if you can just used a different compiler version, that is easiest for now.

gordon-quad, 2022-06-07:
any news on that?
Trying to make a yocto layer for gnubee and I would really like not to fallback to older version of gcc.

It looks like there is mt7621.c in 5.18 kernel, is there a way to backport it?

neilbrown, 2022-06-12:
The offending code was rewritten in 
Commit: cc19db8b312a ("MIPS: ralink: mt7621: do memory detection on KSEG1")
in Linux 5.17.  That patch applies cleanly on 5.15 and compiles.  I haven't tested if it runs.
I've updated my v5.15 branch to be based on 5.15.22 and added this patch.  So you could try building
from there, or just add the patch yourself and try that.

neilbrown, 2022-12-28:
This issue seems to be fixed, so I'm closing.

closed 2022-12-28 by neilbrown