neilbrown/gnubee-tools #38: kernel modules ?

reukiodo, 2023-06-30:
Is there a compile guide to add modules? I'm specifically looking to compile the zram module as it doesn't seem to be included in your image.

So far I've run:
```
cd /home/gnubee
apt -y install make gcc flex bison busybox git bc libssl-dev u-boot-tools unzip rsync mdadm dropbear cryptsetup-bin mtd-utils xfsprogs btrfs-progs lvm2 evtest debootstrap
wget  https://github.com/neilbrown/linux/archive/gnubee/v6.1.zip
unzip v6.1.zip
cd linux-gnubee-v6.1
../gnubee-tools/scripts/gbmake firmware gbpc1-6.1
```
But `find / -type f -name zram.ko` returns nothing. I suppose there is something else needed to edit before it will compile the zram module?

reukiodo, 2023-06-30:
I eventually found https://docs.kernel.org/kbuild/modules.html which leads me to run `make -C /home/gnubee/linux-gnubee-v6.1/ M=/home/gnubee/linux-gnubee-v6.1/drivers/block/zram modules_install` but this just returns
```
make: Entering directory '/home/gnubee/linux-gnubee-v6.1'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

Makefile:741: include/config/auto.conf: No such file or directory
make: *** [Makefile:806: include/config/auto.conf] Error 1
make: Leaving directory '/home/gnubee/linux-gnubee-v6.1'
```
Yet when I run `cd /home/gnubee/linux-gnubee-v6.1 && make oldconfig && make prepare` it asks a ton of questions I have no idea how to answer. If I just accept all the defaults and try again, I'm greeted with:
```
make: Entering directory '/home/gnubee/linux-gnubee-v6.1'
***
*** The present kernel configuration has modules disabled.
*** To use the module feature, please run "make menuconfig" etc.
*** to enable CONFIG_MODULES.
***
make: *** [Makefile:1935: modules_install] Error 1
make: Leaving directory '/home/gnubee/linux-gnubee-v6.1'
```
Which of course doesn't run... `make menuconfig` returns:
```
*
* Unable to find the ncurses package.
* Install ncurses (ncurses-devel or libncurses-dev
* depending on your distribution).
*
* You may also need to install pkg-config to find the
* ncurses installed in a non-default location.
*
make[1]: *** [scripts/kconfig/Makefile:212: scripts/kconfig/mconf-cfg] Error 1
make: *** [Makefile:697: menuconfig] Error 2
```

reukiodo, 2023-06-30:
So after a `apt -y install libncurses-dev` and `make menuconfig` I think I just need to select 'Enable loadable module support'?

reukiodo, 2023-06-30:
nope... same error... what is 'enable CONFIG_MODULES' in menuconfig terms?

reukiodo, 2023-06-30:
```
2023-06-30 06:06:37 root@gnubee:/home/gnubee/linux-gnubee-v6.1# grep CONFIG_MODULES .config
CONFIG_MODULES_USE_ELF_REL=y
CONFIG_MODULES=y
2023-06-30 06:10:55 root@gnubee:/home/gnubee/linux-gnubee-v6.1# 
```
I think the .config is what it is referencing, and `CONFIG_MODULES` is already set to `y` so I don't understand...

reukiodo, 2023-07-01:
So finally got a zram.ko compiled after inserting numerous `CONFIG_MODULES=y` throughout.
```
2023-07-01 17:35:14 root@gnubee:~# find / -type f -name zram.ko
/usr/lib/modules/6.1.1+/kernel/drivers/block/zram/zram.ko
2023-07-01 18:11:33 root@gnubee:~# insmod /usr/lib/modules/6.1.1+/kernel/drivers/block/zram/zram.ko

insmod: ERROR: could not insert module /usr/lib/modules/6.1.1+/kernel/drivers/block/zram/zram.ko: Invalid module format
2023-07-01 18:11:43 root@gnubee:~# cat /proc/version
Linux version 6.1.1+ (neilb@noble) (mipsel-unknown-linux-gnu-gcc (GCC) 7.2.0, GNU ld (GNU Binutils) 2.29.1.20170915) #2 SMP Mon Jan  2 13:00:11 AEDT 2023
2023-07-01 18:12:48 root@gnubee:~# 
```

neilbrown, 2023-07-02:
What I would do in this situation, after running "gbmake firmware" as you did, is to edit "O/.config" and find the line that mentioned ZRAM.  It looked like "# CONFIG_ZRAM is not set"
I just delete that line and then run "gbmake oldconfig".
I get prompted about whether I want ZRAM, and various config options that I might want to choose.
Then "gbmake firmware" should build a new firmware with ZRAM included.

Next time I build a firmware, I'll include ZRAM in it.

reukiodo, 2023-07-02:
So, I guess this means that even if it build the correct zram.ko file, the current kernal cannot load it with a full rebuild and flashing it to SPI?

neilbrown, 2023-07-03:
> So, I guess this means that even if it build the correct zram.ko file, the current kernal cannot load it without a full rebuild and flashing it to SPI?

That isn't true in general.
You need to build the module in largely the same context that the kernel was build, so if it was a different kernel version, that would cause problems.
If you just use gbmake and edit the config file as I described, it *should* work to then "insmod" the zram.ko file.
If it doesn't there are usually kernel messages (see output of "dmesg") that give you a hit of what went wrong.

If that doesn't work you could try:
 echo file kernel/module/main.c +p > /sys/kernel/debug/dynamic_debug/control

and then try the insmod again.  That enables various debug messages in module loading.
Use
 
 echo file kernel/module/main.c -p > /sys/kernel/debug/dynamic_debug/control

to turn the debug messages back off.

reukiodo, 2023-07-03:
I think I messed something up so badly with all the previous edits that I'm restarting from scratch: `rm -rf linux-gnubee-v6.1 gnubee-tools` before:
```
cd /home/gnubee
apt -y install make gcc flex bison busybox git bc libssl-dev u-boot-tools unzip rsync mdadm dropbear cryptsetup-bin mtd-utils xfsprogs btrfs-progs lvm2 evtest debootstrap
wget  https://github.com/neilbrown/linux/archive/gnubee/v6.1.zip
unzip v6.1.zip
cd linux-gnubee-v6.1
../gnubee-tools/scripts/gbmake firmware gbpc2-6.1
```
and will edit the `linux-gnubee-v6.1/O/.config` in the morning when this is done rebuilding. Hopefully then I'll be able to build a compatible zram.ko and test it out.

reukiodo, 2023-07-03:
I think I am still missing something, as I'm not able to get a zram.ko from this:
```
2023-07-03 20:46:59 root@gnubee:/home/gnubee/linux-gnubee-v6.1# grep -i zram O/.config
CONFIG_ZRAM is not set
2023-07-03 20:47:01 root@gnubee:/home/gnubee/linux-gnubee-v6.1# ../gnubee-tools/scripts/gbmake oldconfig
make[1]: Entering directory '/home/gnubee/linux-gnubee-v6.1/O'
  GEN     Makefile
*
* Restart config...
*
*
* Block devices
*
Block devices (BLK_DEV) [Y/n/?] y
  Null test block driver (BLK_DEV_NULL_BLK) [N/m/y/?] n
  Block Device Driver for Micron PCIe SSDs (BLK_DEV_PCIESSD_MTIP32XX) [N/m/y/?] n
  Compressed RAM block device support (ZRAM) [N/m/y/?] (NEW) y
    Default zram compressor
    > 1. lzo-rle (ZRAM_DEF_COMP_LZORLE) (NEW)
      2. zstd (ZRAM_DEF_COMP_ZSTD) (NEW)
      3. lz4 (ZRAM_DEF_COMP_LZ4) (NEW)
      4. lzo (ZRAM_DEF_COMP_LZO) (NEW)
      5. lz4hc (ZRAM_DEF_COMP_LZ4HC) (NEW)
      6. 842 (ZRAM_DEF_COMP_842) (NEW)
    choice[1-6?]: 2
  Write back incompressible or idle page to backing device (ZRAM_WRITEBACK) [N/y/?] (NEW) 
  Track zRam block status (ZRAM_MEMORY_TRACKING) [N/y/?] (NEW) y
  Loopback device support (BLK_DEV_LOOP) [Y/n/m/?] y
    Number of loop devices to pre-create at init time (BLK_DEV_LOOP_MIN_COUNT) [8] 8
  DRBD Distributed Replicated Block Device support (BLK_DEV_DRBD) [N/m/y/?] n
  Network block device support (BLK_DEV_NBD) [N/m/y/?] n
  RAM block device support (BLK_DEV_RAM) [N/m/y/?] n
  Packet writing on CD/DVD media (DEPRECATED) (CDROM_PKTCDVD) [N/m/y/?] n
  ATA over Ethernet support (ATA_OVER_ETH) [N/m/y/?] n
  Rados block device (RBD) (BLK_DEV_RBD) [N/m/y/?] n
  Userspace block driver (Experimental) (BLK_DEV_UBLK) [N/m/y/?] n
*
* Restart config...
*
*
* Support for paging of anonymous memory (swap)
*
Support for paging of anonymous memory (swap) (SWAP) [Y/n/?] y
  Compressed cache for swap pages (ZSWAP) [N/y/?] n
    N:1 compression allocator (zsmalloc) (ZSMALLOC) [Y/?] y
      Export zsmalloc statistics (ZSMALLOC_STAT) [N/y/?] (NEW) y
#
# configuration written to .config
#
make[1]: Leaving directory '/home/gnubee/linux-gnubee-v6.1/O'
2023-07-03 20:56:38 root@gnubee:/home/gnubee/linux-gnubee-v6.1# insmod zram
insmod: ERROR: could not load module zram: No such file or directory
2023-07-03 20:57:13 root@gnubee:/home/gnubee/linux-gnubee-v6.1# insmod zram.ko
insmod: ERROR: could not load module zram.ko: No such file or directory
2023-07-03 20:57:16 root@gnubee:/home/gnubee/linux-gnubee-v6.1# find / -type f -name zram.ko
find: ‘/proc/15749’: No such file or directory
find: ‘/proc/15750’: No such file or directory
2023-07-03 21:02:42 root@gnubee:/home/gnubee/linux-gnubee-v6.1# 
```

I didn't choose most of those options, they were auto-selected, such as `RAM block device support (BLK_DEV_RAM) [N/m/y/?] n` which looks like it may be contributing to why zram.ko wasn't built.

neilbrown, 2023-07-03:
> # grep -i zram O/.config
> CONFIG_ZRAM is not set

It looks like you didn't delete the whole line - just the "#" ??? It seems to have had the desired effect.

It looked like you only ran "gbmake oldconfig" - that just updates the config file.  I said you should then run "gbmake firmware", though you could just run "gbmake ; gbmake modules".
You will need to insmod more than just zram.ko.  At least zsmalloc.ko (found in O/mm) is needed.
(It might be easier just to create a new firmware - but it shouldn't be necessary)

If BLK_DEV_RAM were needed for ZRAM, then it would have been automatically selected.  BLK_DEV_RAM allows you to treat a section of RAM as a block device.  You almost certainly don't need it.

reukiodo, 2023-07-05:
i re-ran `2023-07-05 16:22:05 root@gnubee:/home/gnubee/linux-gnubee-v6.1# ../gnubee-tools/scripts/gbmake firmware gbpc2-6.1` and still don't get a zram.ko out of it (nowhere to be found in a `find / -type f -name zram*`)
It seems to be enabled for compile:
```
2023-07-05 16:19:39 root@gnubee:/home/gnubee/linux-gnubee-v6.1# grep -i zram O/.config
CONFIG_ZRAM=y
# CONFIG_ZRAM_DEF_COMP_LZORLE is not set
CONFIG_ZRAM_DEF_COMP_ZSTD=y
# CONFIG_ZRAM_DEF_COMP_LZ4 is not set
# CONFIG_ZRAM_DEF_COMP_LZO is not set
# CONFIG_ZRAM_DEF_COMP_LZ4HC is not set
# CONFIG_ZRAM_DEF_COMP_842 is not set
CONFIG_ZRAM_DEF_COMP="zstd"
# CONFIG_ZRAM_WRITEBACK is not set
CONFIG_ZRAM_MEMORY_TRACKING=y
2023-07-05 16:20:09 root@gnubee:/home/gnubee/linux-gnubee-v6.1# 
```

i'm trying `gbmake; gbmake modules` now and will update when it's done.

reukiodo, 2023-07-05:
it still seems like I a missing somethinng...
```
2023-07-05 16:22:05 root@gnubee:/home/gnubee/linux-gnubee-v6.1# ../gnubee-tools/scripts/gbmake modules
make[1]: Entering directory '/home/gnubee/linux-gnubee-v6.1/O'
  GEN     Makefile
  CALL    ../scripts/checksyscalls.sh
  GEN     usr/initramfs_data.cpio
  LZMA    usr/initramfs_inc_data
  AS      usr/initramfs_data.o
  AR      usr/built-in.a
  AR      built-in.a
  AR      vmlinux.a
  LD      vmlinux.o
  OBJCOPY modules.builtin.modinfo
  GEN     modules.builtin
  MODPOST Module.symvers
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/arch/mips/crypto/chacha-mips.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/arch/mips/crypto/chacha-mips.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/arch/mips/crypto/poly1305-mips.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/arch/mips/crypto/poly1305-mips.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/842.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/842.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/adiantum.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/adiantum.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/aegis128.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/aegis128.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/aes_ti.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/aes_ti.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/af_alg.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/af_alg.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/algif_aead.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/algif_aead.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/algif_hash.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/algif_hash.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/algif_rng.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/algif_rng.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/algif_skcipher.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/algif_skcipher.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/ansi_cprng.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/ansi_cprng.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/anubis.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/anubis.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/asymmetric_keys/pkcs8_key_parser.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/asymmetric_keys/pkcs8_key_parser.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/async_tx/async_memcpy.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/async_tx/async_memcpy.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/async_tx/async_pq.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/async_tx/async_pq.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/async_tx/async_raid6_recov.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/async_tx/async_tx.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/async_tx/async_xor.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/async_tx/async_xor.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/authenc.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/authenc.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/authencesn.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/authencesn.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/blake2b_generic.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/blake2b_generic.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/blowfish_common.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/blowfish_common.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/blowfish_generic.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/blowfish_generic.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/camellia_generic.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/camellia_generic.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/crypto/cast5_generic.ko
[... cut ...]
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/sound/usb/snd-usb-audio.ko
  INSTALL /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/sound/usb/snd-usbmidi-lib.ko
  STRIP   /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1/kernel/sound/usb/snd-usbmidi-lib.ko
  DEPMOD  /home/gnubee/linux-gnubee-v6.1/initramfs/lib/modules/6.1.1
make[1]: Leaving directory '/home/gnubee/linux-gnubee-v6.1/O'
2023-07-05 16:55:59 root@gnubee:/home/gnubee/linux-gnubee-v6.1# 
```